MakeraInc/CarveraController #3: Bug: Multiple Access Points with the same SSID displayed when attempting to connect to WiFi

jfryman, 2022-12-28:
Hello there. First, thanks for the product. I'm already ecstatic to have it. 

This is a minor bug, but one that has been an annoyance to my ability to connect to WiFi.

## Summary:
In my home, I run multiple Access Points that broadcast the same SSID with Fast Roaming. When attempting to connect to WiFi using the Carvera Controller software, I see each BSSID with the same SSID to connect to.

## Workaround:
In the meantime, I have provisioned a SSID/BSSID combo that is locked to the nearest access point to the device for purposes of connecting the device to the AP with the strongest signal.

## Expected Solution
In most applications, the BSSIDs of each Access Point are collapsed into a single SSID that I connect to. I would expect to see only one SSID when multiple BSSIDs are broadcast, and the machine attempt to connect to the strongest signal presented instead.

wcraigtrader, 2022-12-31:
Pretty sure this is a bug in the firmware, not the controller.

jfryman, 2023-05-24:
Hi there!

Did a fix for this get rolled up into a firmware update?