MakeraInc/CarveraController #9: UI should remember the last WiFi address

wcraigtrader, 2023-05-28:
When connecting to the WiFi, the Controller software should remember the last IP address used when it doesn't automatically detect a Carvera in the local network segment.

BlackholeBun, 2023-06-23:
Seconded. In fact it would be great if it attempted to auto-connect to the last machine on launch. Or at least have an option for it

datacoda, 2023-06-23:
Thirded.  For some reason, my android cannot see the Carvera even while others can.  This would help alleviate the pain.

gskluzacek, 2023-07-12:
I agree with all of the above

redwolf010, 2025-03-12:
Thank you for your suggestion, this issue has been fixed over verion 0.9.10.

kds69, 2025-04-21:
> Thank you for your suggestion, this issue has been fixed over verion 0.9.10.

it seems this is not really fixed. I am running 9.11 and the issue is still there.

Release 0.9.10, indeed says, but I am probaly not the only one but I still have to put again and again the IP address of the machine.
BTW what is the machine name? When my Carvera Air has joine my network, no host name was reported at all. So maybe that is the is primary issue?

`3. Optimizing: The 'Recent connect' button has been optimized to automatically search for a new IP based on the machine name when the IP is changed.`