MakeraInc/CarveraController #20: "CarveraController" is damaged and can't be opened on MacOS 13.6.3

eric2777, 2024-01-24:
After fresh install of CarveraController v0.9.6 binary downloaded from this repository, macOS generated the following error
[image]

MakeraInc, 2024-01-24:
Please see the solution at:
[https://github.com/MakeraInc/CarveraController?tab=readme-ov-file#macosx-installation](https://github.com/MakeraInc/CarveraController?tab=readme-ov-file#macosx-installation)

eric2777, 2024-02-03:
Thanks for the information. After running attr, I don't get the previous error but the app doesn't start and crash silently.
I ran the other terminal command as explained and I get the following error:

[image]

eric2777, 2024-02-03:
My bad, I just downloaded the latest build that was released 3 days ago, and now I can launch the app. So this issue can be closed now. thanks

eric2777, 2024-02-04:
fixed in latest version

closed 2024-02-04 by eric2777