MakeraInc/CarveraController #22: Machine loses home and jogs at a 10x scaling factor after running a gcode program

AlexRichBT, 2024-03-22:
FW version 0.9.7, Windows PC Controller v 0.9.6, Fusion post version 43151:

Any time I run a program and the program finishes, or if I stop the program in the middle – the controller malfunctions.  It loses its home position, and some sort of scaling issue occurs where all moves in X, Y, and Z are scaled 10x.  So a jog of 10mm is 100mm, a jog of 1mm is 10mm, etc.  I don’t know why that would be.  Essentially I have to reset the machine every time I finish or stop a program, I have to re-find my work origin, etc.  The machine is not usable with this controller.

MakeraInc, 2024-03-23:
Hi, could you please share the g-code file you are using? I am guessing there is an inch-switching command: G20 which the Carvera doesn't support for now, please use G21 (Metric) instead.  Which CAM software you are using, please check Metric mode.

AlexRichBT, 2024-03-23:
Ok that has to be it, yes I was in inch mode.  You guys should put a note about that on the Fusion 360 setup instructions if it's not there already.  It would be great if the controller did some error checking of the gcode file, if G20 is used in the gcode then flag it with a popup error - this cost me a full day of confusion!  I don't know if you check for moves outside of the machine envelope but if you don't then that is a necessary check also.
 
Alex

-Alex

AlexRichBT, 2024-03-23:
I would suggest in the meantime if you can't add a popup error to the controller software, just add a warning in the fusion 360 post that will put a big comment at the top of the gcode.  I did it on my post by adding this snippet at line 371:

```
if(unit == IN) {
    writeComment("************** WARNING **************\nCARVERA ONLY WORKS IN MM MODE, G21.\nTHIS PROGRAM IS IN INCH MODE , G20\n*************************************");
  }
```