MakeraInc/CarveraController #29: Android version can only start jobs at Anchor 1 or Anchor 2

jwr, 2024-10-15:
When setting the work origin in the Android version of the Carvera controller, there is no "current position" option. Consequently, one cannot probe the work origin independently and have the machine start at the probed work origin. 

Please add the "Current Position" option to the "Set Work Origin" dialog in the Android version. Ideally, please also add the option not to set the work origin at all, as reported in #28.

redwolf010, 2024-10-23:
[image]
The Android version already includes the "Current Position" option in the "Set Work Origin" dialog

redwolf010, 2025-03-27:
The Android version already includes the "Current Position" option in the "Set Work Origin" dialog

Action: This issue can be closed.

Additional Notes:

If any related problems reappear in future testing, please reopen the issue or log a new report.
For further optimizations or feature requests, feel free to submit feedback.
Thank you for your contribution!