MakeraInc/CarveraController #31: Need 64bit support for recent Google Pixel devices

stl314159, 2025-01-21:


Bostwickenator, 2025-02-09:
Same issue here

[image]

dzhang97, 2025-03-08:
FYI v0.9.9 works, the apk contains the 64b libraries even thought v0.9.10 doesn't

redwolf010, 2025-03-27:
After reviewing v0.9.11, the reported issues have been verified as resolved. The fixes and optimizations mentioned in the changelog (including arc machining accuracy, tool number handling, lighting control, multi-language support, and Android 64-bit compatibility) are now functioning as intended.

Action: This issue can be closed.

Additional Notes:

If any related problems reappear in future testing, please reopen the issue or log a new report.
For further optimizations or feature requests, feel free to submit feedback.
Thank you for your contribution!